riscv/riscv-fast-interrupt #445: Note on relative priority of platform interrupts should cite sources

mark-honman, 2025-02-04:
In section 3.1.2 of the CLIC spec we have this note:

"The highest numbered interrupt wins in a tie (when privilege mode, level and priority are all identical). This is the same as in CLINT interrupt mode, but different than the PLIC."

I've not been able to find where the CLINT interrupt mode is defined to have highest numbered interrupt winning in a tie. 
The Privileged ISA manual (20240411) has a note in section 3.1.9 which states:
"The platform-specific machine-level interrupt sources in bits 16 and above have platformspecific priority, but are typically chosen to have the highest service priority to support very fast local vectored interrupts." 

The statement in the CLIC spec appears to be in conflict with this.

christian-herber-nxp, 2025-02-04:
I think the note is unnecessary. There is a ton of interrupt controllers specs in RISC-V. Makes no sense that every interrupt controller spec states how it does prioritization differently. This is bound to be incomplete and redundant.

mark-honman, 2025-02-07:
Yes, that's a good idea... reduces confusion!

smatherd, 2025-04-07:
currently removed do to #464.  ok to close?

mark-honman, 2025-04-07:
Yes, clear to close. 
I like the latest wording in the spec
"priority is determined using the AIA iprio rule"

closed 2025-04-07 by mark-honman