riscv/riscv-fast-interrupt #446: mtvec is not backwards compatible due to alignment

Timmmm, 2025-02-04:
It's pretty annoying that existing code that uses `mtvec` can get broken just by running it on a CLIC core because CLIC increases the alignment requirement from 2^2 to 2^6 bytes. The failure mode can also be pretty bad - it can just jump to a random incorrect address.

Why not put the `submode` in the low bits of `xtvt`? Or just remove it entirely since it isn't even used.

jnk0le, 2025-02-04:
`submode` is active only with `mode` set to 11. There is no alignment requirement in "legacy" mode.

> Systems implementing both CLIC and CLINT mode may, but are not required to, limit
alignment of mtvec to 64-byte boundaries in both modes.

smatherd, 2025-04-07:
#464 update currently removes submode and 2^6 alignment.  ok to close for now?

closed 2025-04-07 by christian-herber-nxp