riscv/riscv-fast-interrupt #450: Value of clicintip when switching to edge-triggered mode should be defined

Timmmm, 2025-02-24:
> The value in the clicintip[i] is undefined when switching from level-sensitive mode to edge-triggered mode in clicintattr[i].

This seems unnecessarily timid. Is there a good reason not to zero it?

smatherd, 2025-02-27:
When clicintattr and clicintip were in the same memory mapped word, it was low burden on SW to write both at the same time.  Still seems like a low SW burden with the indirect CSRs since it only happens during configuration, right?

platform specifications or company policy could zero it if they like.  clic spec just doesn't mandate it.

Here is some history:
https://github.com/riscv/riscv-fast-interrupt/issues/153
https://github.com/riscv/riscv-fast-interrupt/blob/master/minutes/20210427-Fast-Interrupts-minutes.pdf

20210427 minutes:  (sorry there isn't much detail on the reasoning)
#152 (2nd part of #149) - changing from edge to level, what happens to pending? Only the 
current values of clicintip affect the interrupt processing. Closed with spec update clarifying 
behavior of the clicintip bit.
#153 (based on 2nd part of #149 continued) - what happens to clicintip value when switching 
from edge to level and back to edge. The state in clicintip is undefined in this case. Closed with 
spec update.

Timmmm, 2025-02-27:
The issue isn't that it's hard for *correctly written* software to do it. Leaving it as undefined behaviour has significant costs for software and verification:

* Testing software is now *much* hardware because you can no longer be tested on a single implementation.
* It is easy to write software that *accidentally* works on one implementation but would break on a different one. That's bad.
* Silicon verification is significantly harder because you no longer have a single possible outcome, but a range of legal outcomes.

Those things have to be traded off against the advantages for hardware designers & architects:

* Allows future spec changes more easily.
* Less to think about when writing specs ("eh who knows; leave it undefined")
* More flexibility for hardware designs.

But I think the hardware costs in this case are trivial so the question is not "is it easy for software to deal with?", it's "is it easy enough for hardware to deal with that we don't need to impose undefined behaviour costs on software/verif?".

I can't see any reason this wouldn't be trivial to implement in hardware, but I'm not a design expert...

smatherd, 2025-02-27:
Before SW enables any edge-triggered interrupt, wouldn't it clear clicintip first anyway to make sure there wasn't garbage there? (e.g. an old pulse happened a long time ago and was captured by the edge-triggered logic but isn't relevant anymore.    I don't think you gain anything by requiring that it be 0 when switching to edge-triggered mode since that all is done in setup phase before interrupts would be enabled.

There's also the posedge vs. negedge setting.  I assume when switching that you'd want clicintip defined as well?

christian-herber-nxp, 2025-03-05:
@Timmmm due to other issues, I want to introduce the possibility to optionally have the pending bits writable for level based interrupts.
In this case, we would end up with a clearly defined behavior when switching. Please review the PR and let me know if that works for you.

smatherd, 2025-03-07:
Reading AIA spec, it looks like APLIC is similarly undefined after changing the edge/level configuration:
> An edge-sensitive source can be configured to recognize an incoming interrupt on either a rising edge (low-to-high transition) or a falling edge (high-to-low transition). When configured for a falling edge (mode Edge0), the source is said to be inverted.
...
> Any write to a sourcecfg register might (or might not) cause the corresponding interrupt-pending bit to be set to one if the rectified input value is high (= 1) under the new source mode.

christian-herber-nxp, 2025-04-07:
stick to the precedent in AIA.

closed 2025-04-07 by christian-herber-nxp