riscv/riscv-fast-interrupt #451: Value of xpil in case of a synchronous exception

christian-herber-nxp, 2025-02-25:
I tried to find this, but couldn't figure out what the value of xpil would be for an exception.
The text states that xpil is set on a trap, but the name suggests it is only interrupt related.

smatherd, 2025-02-27:
We tried to use the same text as in the priv spec to describe xpil behavior:

As stated in the RISC-V privilege specification, when a trap is taken from privilege mode y into privilege mode x, xPIE is set to the value of xIE; xIE is set to 0; and xPP is set to y. xepc is written with the virtual address of the instruction that was interrupted or that encountered the exception. 

This is what was added in the clic spec:
Additionally in CLIC mode, interrupt level (xpil) is set to xintstatus.xil and xcause.exccode is written with a code indicating the event (the id of the interrupt or exception code) that caused the trap.

So I interpret that to mean xpil is set to xintstatus.xil on all traps, exceptions and interrupts.

christian-herber-nxp, 2025-02-27:
Sorry, I accidentally asked the wrong question. My question was supposed to be what is the value of xil in that case?

christian-herber-nxp, 2025-02-27:
found it:

https://github.com/riscv/riscv-fast-interrupt/blob/master/src/clic.adoc#3104-synchronous-exception-handling.

Was it intentional that the conditional scratch swap did not work for exceptions? Because xpil will always be xil. 
What I see in Zephyr is that exceptions use the same stack as interrupts.

closed 2025-04-14 by christian-herber-nxp