riscv/riscv-fast-interrupt #469: intstatus is creating a virtualization loophole

christian-herber-nxp, 2025-04-01:
the privilege mode should not be exposed at a hart to avoid virtualization holes.
Currently, you can detect this through intstatus.

if you are seeing a non-zero level in S-mode, and the M-mode level is 0, you can be sure to be in S-mode.
if mil is non-zero, you are in M-mode.

As this is a non-essential feature, I would recommend to remove.

smatherd, 2025-04-05:
> if you are seeing a non-zero level in S-mode, and the M-mode level is 0, you can be sure to be in S-mode.

I think that if you were in an s-mode handler and you took an exception handled in m-mode, you would see non-zero level in s-mode and m-mode level 0.   but you are in m-mode, not s-mode.

> if mil is non-zero, you are in m-mode.

right, you can only access mintstatus.mil in m-mode.  otherwise you get an exception.  so I don't think it is a virtualization hole.  It's been a useful CSR for FW debug.

christian-herber-nxp, 2025-04-07:
as this for debug, we can make it an M-mode register

commit fe478a96 (2025-04-09): Fixes #469: Removing S mode debug register to prevent virtualization loophole
commit e3d51d7c (2025-04-10): Fixes #469: Removing S mode debug register to prevent virtualization loophole
closed 2025-04-10 by christian-herber-nxp