riscv/riscv-fast-interrupt #470: Redundancy in description of interrupt enable

christian-herber-nxp, 2025-04-01:
> Within a privilege mode `*_x_*`, if the associated global
> interrupt-enable {ie} is clear, then no interrupts will be taken in
> that privilege mode, but a pending-enabled interrupt in a higher
> privilege mode will preempt current execution.  If {ie} is set, then
> pending-enabled interrupts at a higher interrupt level in the same
> privilege mode will preempt current execution and run the interrupt
> handler for the higher interrupt level.

> NOTE: Existing RISC-V interrupt behavior is retained, where incoming
> interrupts for a higher privilege mode can preempt an active interrupt
> handler running in a lower privilege mode, regardless of global
> interrupt enable in lower privilege mode.

I believe only one needs to be kept. The note seems to be stating the same and is more fitting, as this is already defined behavior in M/S.

smatherd, 2025-04-03:
This is the key difference between clic and clint:

> If {ie} is set, then
pending-enabled interrupts at a **higher interrupt level** in the same
privilege mode will preempt current execution

in clint, if {ie} is set, any pending-enabled interrupt will preempt current execution.  The note does not mention that.

maybe change the text to combine them?
>Within a privilege mode *_x_*, if the associated global
interrupt-enable {ie} is clear, then no interrupts will be taken in
that privilege mode, but a pending-enabled interrupt in a higher
privilege mode will preempt current execution regardless of global interrupt enables, like existing RISC-V interrupt behavior. If {ie} is set, then
pending-enabled interrupts at a higher interrupt level in the same
privilege mode will preempt current execution and run the interrupt
handler for the higher interrupt level.

christian-herber-nxp, 2025-04-03:
My suggestion:

The existing interrupt behavior is retained, with the following addition:
Interrupts are only considered pending, if their pending bit is set and their level is greater-equal the current effective interrupt level.

Note: The effective term is essential here to also capture the threshold behavior.

closed 2025-04-14 by christian-herber-nxp