riscv/riscv-fast-interrupt #494: Zifenci extension mandated

christian-herber-nxp, 2025-04-08:
> Memory writes to the vector table require an instruction barrier (_fence.i_) to guarantee that they are visible to the instruction fetch.

this piece of text implies a hard dependency towards the Zifencei extension. Is this truly the case? If so, we be explicit about it.

christian-herber-nxp, 2025-04-08:
after giving this some thought, I think just having interrupts disabled while making these changes is enough.

smatherd, 2025-04-08:
This is in the zcmt spec:
> Memory writes to the jump vector table require an instruction barrier (fence.i) to guarantee that they
are visible to the instruction fetch.
Multiple contexts may have different jump vector tables. JVT may be switched between them without
an instruction barrier if the tables have not been updated in memory since the last fence.i.

hw vector would have the same considerations.  Not really mandated, more a helpful note as a reminder that if the memory write is a cached write (in dcache) and instruction fetch table is allowed to be in a separate icache, fence.i would be needed to make sure the update would be seen by instruction fetch.  if the vector table was held in uncached local memory, fence.i wouldn't be needed.  but since this statement is in zcmt, it makes sense for clic to have it too.

christian-herber-nxp, 2025-04-08:
got it, closed.

closed 2025-04-08 by christian-herber-nxp