riscv/riscv-fast-interrupt #498: Handling of faults encountered during tpush and tpopxret

Silabs-ArjanB, 2025-04-09:
Section 8.5. Fault handling states the following:

> According the the Machine Level ISA, trap handlers must be designed to not cause exceptions during their critical section. This applies to the tpush and tpopxret instruction. Therefore, by design, it must be guaranteed that no faults occur during the execution of these instructions.

Can you please provide the specific quote in the Machine Level ISA about this or point me to the relevant section?

Although I agree that faults should be avoided, I don't think that this can be guaranteed in general and therefore I think it is important to specify the required behavior in case a fault occurs nonetheless (e.g. by triggering a double fault, etc.)

christian-herber-nxp, 2025-04-09:
I would have to search through the priv manual. It is somewhere in the M mode definition.
However, I agree that the only real solution is double trap.
As I have meanwhile added interaction with double trap handling to push/pop, I need to adjust this paragraph.

christian-herber-nxp, 2025-04-09:
Made small adjustment in the linked PR.

closed 2025-04-14 by christian-herber-nxp