riscv/riscv-fast-interrupt #500: Why does each interrupt i have both a cliciprio[i] and clicintlvl[i]?

Silabs-ArjanB, 2025-04-09:
Section 2.6.2. cliciprio[i] defines a cliciprio for each interrupt. Section 3.1. CLIC Interrupt Input Control (clicintlvl) defines a clicintlvl[i] per interrupt. 

In contrast the AIA specification does only have iprio[i] for the non-external interrupts. The iprio value for the machine-level external interrupt must be read-only zero. 

What is the reason for the possibility to annotate both level and prio for external interrupts (as this was not possible for AIA)?

And an even more fundamental question: Why bother about cliciprio at all? Instead all interrupt sources could be hooked up to an interrupt controller that just performs prioritization based on clicintlvl so that we would not have to deal with the two step prioritization caused by the legacy CLINT architecture.

christian-herber-nxp, 2025-04-09:
I am not necessarily happy with how the level and prio is currently speced. 
Arm has an equivalent split between prio and level called group priority and subpriority.

One reason why having this split can be useful is to limit the worst case interrupt stack size (which is the sum of the largest possible stacks per level). More levels means bigger stack.
Do you agree this makes sense?

Silabs-ArjanB, 2025-04-09:
In a previous version of the CLIC spec (version v0.9, 2024-06-28: Draft) there was the concept of 'interrupt priority' and 'interrupt level'. In that spec (only) level affects preemption and priority could be used to prioritize between interrupts at the same level. In that CLIC spec priority and level were **both** used to pick the most important interrupt to present to the core. I was completely fine with that proposal. That proposal was not related to AIA iprio in any way.

The current CLIC spec seems to have changed that in a way that only the level is used to pick the most important interrupt to present to the core. cliciprio gets factored in later, which is weird as then it might be too late (at only 1 external interrupt will be presented to the core and that might not have the highest cliciprio for a given level).

I agree it makes sense to limit the number of levels (levels of preemption), but I am really puzzled at why cliciprio was introduced that seems to match AIA iprio for some interrupts. If you want something like this then you need three concepts I think: level to be used to rank external interrupts, prio to rank external interrupts, prio for non external interrupts to rank against the one external interrupt presented to the core. Again, why use AIA and AIA iprio at all, it just seems to complicate things.

christian-herber-nxp, 2025-04-14:
TG Meeting:
- Specify if a partial implementation of lvl implements LSBs or MSBs.

christian-herber-nxp, 2025-06-02:
outdated, went back to one share prio configuration that can be split into part that preempts and the rest.

closed 2025-06-02 by christian-herber-nxp