riscv/riscv-fast-interrupt #516: Clarification on Appendix F: CLIC Interrupt ID ordering recommendations?

snps-straus, 2025-04-30:
_Appendix F: CLIC Interrupt ID ordering recommendations_ includes one recommendation that explicitly references a CLIC that also supports CLINT mode (_F.1. CLIC-mode interrupt-map for systems retaining interrupt ID compatible with CLINT mode_)

1. Can you clarify whether the other recommended interrupt-maps (F.2 thru F.4) would be applicable in a **CLIC that supports CLINT mode**, or are those only intended to be used for CLIC-only implementations? 

2. And if so, would some of the "local"/"other" interrupt inputs be expected to be mapped to the bits in the mip/mie CSRs that are designated for platform use (bits 16 and above) when in CLINT mode?

Thanks

Jim Straus
Synopsys

christian-herber-nxp, 2025-05-05:
this appendix was removed. please review the latest specification (clic.adoc) in the repository

closed 2025-05-05 by christian-herber-nxp

smatherd, 2025-05-05:
The appendix is currently removed from the latest version of the clic spec.  We are trying to simplify the clic spec and match as closely as possible to ratified specs.  Recommended interrupt-maps really belong in platform specs.  Anything CLINT mode I would recommend referring to the ratified priv spec and the AIA spec, e.g. AIA spec, Table 8.  The standard major interrupt codes.