riscv/riscv-fast-interrupt #522: Add mechanism analogous to claimi for SW vectored interrupts

christian-herber-nxp, 2025-05-06:
AIA APLIC has a claimi register with the following definition:

> Register claimi has the same value as topi. When this value is not zero, reading claimi has the
> simultaneous side effect of clearing the pending bit for the reported interrupt identity, if possible.
> See Section 4.7 for exactly when the pending bit is cleared by a read of claimi.
> A read from claimi that returns a value of zero has the simultaneous side effect of setting the
> iforce register to zero.
> Writes to claimi are ignored.

We should the same at the hart level. This does somewhat imply having a topi register, which comes from AIA.
However, I believe that we do really need this if we want to have SW-vectored interrupts.

We could reuse the m/snxti encoding for this new m/sclami register.

@smatherd: What do you say: Mandate AIA or not?

mark-honman, 2025-05-06:
If we were to continue with mnxti style encoding then it would be confusing to use register names that echo AIA. 
AIA style names & encodings belong together - this would be a viable option. 

Otherwise, for the specific task of looking up a VT entry, a single register would be sufficient - m/sclaimtvti ?
Containing the address of a table entry computed from m/stvt (as is the case of mnxti) with claimi like behaviour, none of the mnxti funny stuff.

Either option is fine; starting with MTVI and an AIA-encoded m/sclaimi to compute an address in the vector table is an extra 3 instructions vs. reading a pre-computed address from m/sclaimtvti.  Not a significant increase in overhead in the context of SW vectoring.

christian-herber-nxp, 2025-05-06:
by reuse the encoding, I just meant use the CRS number. These are allocated by the ARC, and we have the allocation for mnxti which is now unused.