riscv/riscv-fast-interrupt #525: Proposal: Remove levels to be more consistent with existing specs and terminology

christian-herber-nxp, 2025-05-07:
There were some changes introduced when rebasing on top of AIA that I do not agree with.
Level was given its own register, in addition to priority. 

I would propose the following: work with iprio as defined in AIA. This is the priority used for selecting the next interrupt and no changes are needed to that.
In addition, have a CSR which defines a number lsbs of iprio that are masked for deciding preemption. This would be similar to what was in the original spec. We can call this a preemption priority. 
The preemption priority can then replace level.
This would imply a smaller level has higher priority.

christian-herber-nxp, 2025-05-12:
i do not think any of these issues are fundamental issue of the method of defining a partition of a priority that participates in the preemption logic.
Cortex-M has done this for more than 20 years ever since Cortex-M3. I cannot possibly imagine something not right with this if properly specified.

mark-honman, 2025-05-14:
Certainly levels need to work more like the reworked AIA style priorities. Current inconsistencies are:
* numerically higher level preempts but numerically lower priority is preferred. 
* implemented lvl bits are at the MS end of cliclevel, implemented prio bits at the LS end of clicprio
* hard-to-explain mapping from level-bit values to level in MINTTHRESH remains (but is no longer in the spec).

I'm okay with either a separate level register, or a level-mask solution but if the CLIC has level registers then their use-model should be consistent with the priority regs. 

But... the mask approach is seems cleaner to me, and means that the balance of levels and priorities is configurable at run-time (I guess we would then be calling levels them "priority bands" or "priority levels"). It could probably eliminate the need for MINTTHRESH - adjusting the mask would have a similar effect.  

It's easy for me to say because I'm not doing the work, but it seems to me that it would be useful to see how the mask concept works out - if it becomes complicated to specify well, that would be a red flag.

closed 2025-06-02 by christian-herber-nxp