riscv/riscv-fast-interrupt #553: mtvec bitfields are swapped

bgrayson-mips, 2025-06-20:
The bitfields for `mtvec` have MODE at the MSBs, and BASE at the LSBs by mistake (one of our DV engineers noticed this, I don't deserve credit!)

Source is at https://github.com/riscv/riscv-fast-interrupt/blob/33028d8b536a814e70335355f519900b7097f5be/src/images/wavedrom/xtvec.edn#L9, in case the image doesn't come across well from the released rendering in the PDF.

[image]

christian-herber-nxp, 2025-06-23:
@smatherd: do you know what is the origin and purpose of submode? It is not mentioned in the spec.

smatherd, 2025-06-23:
They just didn’t want to use up all the mtvec [1:0] encodings.   So submode
allowed for future extensions.

christian-herber-nxp, 2025-06-23:
but 
`OBASE  = xtvec[XLEN-1:2]<<2`
meaning the submode is part of the base address for exceptions.
That does not make sense to me. Maybe something was messed up in all the rework?

There is still one encoding of mode which is reserved.

closed 2025-06-26 by christian-herber-nxp