riscv/riscv-fast-interrupt #566: ACLIC spec does not define how external interrupt vector table is indexed

evgeniy-paltsev, 2025-08-11:
ACLIC spec mentions that in "Interrupt vector table mode" PC is set to 
```
M[VTBASE+XLEN/8*exccode] & VTMASK
```
on interrupt. Spec define what is VTBASE but does not define what is `exccode` for external interrupts.

The `exccode` term is not used in AIA or RISC-V Privileged ISA spec. The only mention in ACLIC spec of what `exccode` may be is in the `tpush` instruction description (and only for synchronous exception):

[image]

The `xcause` is same for all external interrupts, so it cannot be used for indexing.

I expect that Interrupt identity (source number) of the external interrupt is intended to be used here, but the spec should describe it explicitly.

----
ACLIC document version: _Version dev, 2025-07-14: Draft_

christian-herber-nxp, 2025-08-11:
please review https://github.com/riscv/riscv-fast-interrupt/pull/570

closed 2025-08-11 by christian-herber-nxp

evgeniy-paltsev, 2025-08-13:
Hi @christian-herber-nxp,

the updated description is more clear, thanks!