riscv/riscv-fast-interrupt #572: Unnecessary dependency on Smaia/Ssaia

reosdev, 2025-08-11:
As per RISC-V AIA, the APLIC domain is the interrupt configuration interface for the group of physical cores running the same privilege level. 
The core's privilege may or may not have permissions to access APLIC domain configuration and/or receive the interrupts delivered by messages or directly (mutually exclusive at domain level). 

APLIC direct interrupt delivery is intended to supply the cores that are not aware of AIA. In such case there is no AIA specific support is required at the core side except the ability to receive single wired interrupt. This means that there is no dependency on Smaia and Ssaia for APLIC/IDC based configurations.  

To control the interrupts APLIC provides the IDC (interrupt delivery control) MMIO interface individual for the privilege level (M or S, but not VS) for each physical core.  In case of single core, the APLIC may have two domains (M and S) and provide IDC interfaces for M-mode and S-mode privileges (direct delivery to VS-mode is not assumed by AIA). 

Mixing of APLIC domain configuration, APLIC IDC, and IMSIC interfaces (as it is documented now) looks incorrect and confusing. Instead it would be reasonable to reuse only APLIC domain configuration and IDC structure as is, improving by migration to iCSRs instead of MMIO. IMSIC interface should not be inherited/exposed in such case and dependency on Ssaia/Smaia looks as unnecessary overhead for single core.

christian-herber-nxp, 2025-08-11:
> Mixing of APLIC domain configuration, APLIC IDC, and IMSIC interfaces (as it is documented now) looks incorrect and confusing. Instead it would be reasonable to reuse only APLIC domain configuration [...]

APLIC with IDC is more costly, as you outlined in your calculation earlier on the mailing list. You would need one prio tree per domain, while with the ACLIC solution, a single prio tree that includes the mode bits is acceptable. Also ACLIC is run-time SW compatible with IMSIC register files.

reosdev, 2025-08-11:
APLIC with IDC is full equivalent to the ACLIC with mode bit. 

In case of single core, the M-domain interrupts have mode bit fixed to 0.   The S-mode domain interrupts have mode bit fixed to 1. Arbitration can be done with consideration of this extra bit. 

Where do you see the difference?

christian-herber-nxp, 2025-08-12:
> #Also ACLIC is run-time SW compatible with IMSIC register files.
> 
> There is very limited compatibility between ACLIC and AIA IMSIC:
> 
> * ACLIC register layout is not compatible with AIA IMSIC interface (different layout and functionality).
> * ACLIC interrupt handling significantly differs compared to the one defined by Smaia/Ssaia (Software that is not aware of ACLIC cannot use it).
> * ACLIC interrupt handling is not defined for cases when vectorization is not enabled or partially enabled. For now it looks incompatible.
> * ACLIC interrupt handling mode is not ISA compatible ({x}tvec.MODE filed is ignored or not described).
> * ACLIC interface to P&E bits is not compatible (indexed by interrupt number instead of prio). Now it is even separate for set and clear bits.
> * ACLIC major interrupts do not conform to ISA as missing mandatory major external interrupts and interrupt injection feature.
> * ACLIC way to access iCSRs is incompatible with AIA which limits usage of indirect data registers to one.

I think you are mixing extensions. If I only look at Smclic extension, I believe it is run-time compatible with Smaia.
I.e. the same handler could work for both Smaia and Smclic external and internal interrupts. The only interaction needed is with topi and topei.
Specifically, the described algorithm in "Interrupt delivery and handling" works without a change with Smclic.

Of course, if you add additional extensions like the preemption or the interrupt vector tables, then there are changes needed.

reosdev, 2025-08-12:
Lets postpone the compatibility for a bit and conclude on dependencies. Current ACLIC document "Chapter 2. Advanced Core Local Interrupt Controller" says:

    The Smaclic extension depends on the Smaia extension. The Ssaclic extension depends on the Ssaia extension.

What kind of dependencies on AIA are required for ACLIC? Why standard interrupt architecture (as per RISC-V Privileged ISA) is not enough?

christian-herber-nxp, 2025-08-12:
> What kind of dependencies on AIA are required for ACLIC? Why standard interrupt architecture (as per RISC-V Privileged ISA) is not enough?

It depends on the following CSRs:
- eidelivery, eithreshold, eipk, eiek
- xtopei
- xtopi

This dependency also pulls in the change of width for mie/mip, etc.
Additionally, this introduces the iprio array for major interrupts.