riscv/riscv-fast-interrupt #573: Incorrect mixture of AIA interfaces

reosdev, 2025-08-11:
RISC-V AIA specifies two interrupt delivery mechanisms: 

1. MSI for the cores that support AIA. In this case the core side shall implement IMSIC file(s) with associated CSR-based interface. This is not supported by ACLIC.
2. Direct delivery to the cores that are not aware of AIA (incl. none RISC-V cores). There is no AIA functionality required at the core side except ability to handle single wired external interrupt. For such cores the APLIC provides IDC (interrupt delivery control) MMIO structure in addition to the domain configuration interface. 

IDC MMIO structure is similar but not equal to the IMSIC CSR interface. ACLIC document describes mixture of AIA APLIC domain configuration, IDC and IMSIC interfaces. Short summary in the following table:

    Reg name        IMSIC CSRs                  IDC MMIO structure      ACLIC  
    
    topi            major id/prio               external id/prio        IMSIC-style  
    topei           external eiid               n/a                     alias to IDC.topi, adds claim, wrong and confusing   
    claimi          n/a                         ext claim               removed, reason not clear  
    eithreshold     eiid/prio threshold         n/a                     alias to IDC ithreshold, wrong usage  
    ithreshold      n/a                         id threshold            aliased, shall be used instead of eithreshold  
    iedelivery      delivery on/off/mode        n/a                     IMSIC-style, wrong and confusing  
    idelivery       n/a                         delivery on/off         removed, reason unclear  
    eie             enable                      n/a                     IMSIC-style alias to APLIC domain configuration, index by ID (instead EIID) is wrong and confusing   
    eip             pending                     n/a                     IMSIC-style alias to APLIC domain configuration, index by ID (instead of EIID) is wrong and confusing   

ACLIC specification mixes the IMSIC, APLIC IDC and domain configuration interfaces in part of registers names and functions. From one side it looks like APLIC IDC, from other side it mimics to IMSIC controller.  

The suggestion is to either adopt APLIC domain configuration and IDC interface or switch to IMSIC-based core side controller.

christian-herber-nxp, 2025-08-15:
CSR instructions are somewhat different the load/stores.
As an example, you cannot map IDC claimi to a CSR, as it has a side effect on read, which is disallowed on CSRs.

How does you proposal expose the interrupt enable bits and the pending bits?

As in the other issue, I think mixing the redirection of major interrupts to external interrupts into this is unnecessary. 
I think we should just discuss the interface.

> "Preemption configuration" (xipreemptcfg) and "previous interrupt context" (xpinstatus) can be moved to iCSRs (not required for interrupt handlers, no need of CSR efficiency).

This can be discussed, but I would like to see a separate issue on that. As the spec is modular, fast access to this is needed if you do not implement trap push/pop.

reosdev, 2025-08-15:
> CSR instructions are somewhat different the load/stores.
> As an example, you cannot map IDC claimi to a CSR, as it has a side effect on read, which is disallowed on CSRs.

Do not understand this comment. The proposal above assumes iCSR-mapped interface. Actually the only xintstatus (from current ACLIC specification) cannot be easilly moved to iCSRs. 

> As in the other issue, I think mixing the redirection of major interrupts to external interrupts into this is unnecessary.
> I think we should just discuss the interface.

Don't you agree that implementing of iprio array and 32-elements vector tables for major interrupts is expensive for low-area configurations? 

> This can be discussed, but I would like to see a separate issue on that. As the spec is modular, fast access to this is needed if you do not implement trap push/pop.

Lets first agree on overall interface look.

christian-herber-nxp, 2025-08-15:
> Do not understand this comment. The proposal above assumes iCSR-mapped interface. Actually the only xintstatus (from current ACLIC specification) cannot be easilly moved to iCSRs.

claimi in the APLIC is defined as 

```
When this value is not zero, reading claimi has the
simultaneous side effect of clearing the pending bit for the reported interrupt identity, if possible.
```

You cannot expose such a register to an indirect CSR access.

As there is already an xtopei register defined, I think it will be very hard to justify adding another such mechanism.

reosdev, 2025-08-15:
> You cannot expose such a register to an indirect CSR access.

I don't see the problem. If this works in case of MMIO (as used by APLIC), then why the iCSR mapped version may have trouble?

reosdev, 2025-08-15:
According to your citation of AIA/APLIC:

> When this value is not zero, reading claimi has the
> simultaneous side effect of clearing the pending bit for the reported interrupt identity, **if possible**.

I still see no problem there. Immediate effect is not required, just predictable. This claim interface is required for non-preemptive handling, so there always enough cycles before next interrupt may be taken.