riscv/riscv-fast-interrupt #574: Inefficient vectorization

reosdev, 2025-08-11:
ACLIC architecture allows exceptions, major (local), and external interrupts to be vectorized. This is specified in a very inefficient way (in terms of HW resources utilization). In case of M and S levels are both configured, at least half of interrupt vector table (IVT) at each privilege is not usable (but the size cannot be reduced if indexed by interrupt number). 

Reserving memory for IVT (assume extra 32 bits memory per interrupt for each privilege other than M-mode) eliminates the benefit of removing IMSIC files (2 bits per interrupt). Rough estimation: 

    1 IMSIC P/E flip-flop bit ~ 10 IVT memory bits

christian-herber-nxp, 2025-08-11:
> ACLIC architecture allows exceptions, major (local), and external interrupts to be vectorized. This is specified in a very inefficient way (in terms of HW resources utilization). In case of M and S [...]

No. mivt and sivt can have the same value. In fact, as sivt is WARL, it may be a read only alias of mivt. In fact, I would consider this a common use case. Only if you need the PMP to separate the vector table entries to the modes do you need a different value. I do not think that there is a need to do so. The ivt will be execute only privilege after initial configuration.

christian-herber-nxp, 2025-08-11:
can you provide further details why this would be an issue? E.g. what would be the fault scenario that is undetected?
For security, the vector tables are not critical in my opinion, as they only lead to handlers, which are PMP protected.
So even if you can trick the CPU to fetch the wrong vector table entry, this does not take you anywhere, but straight to an exception handler.

christian-herber-nxp, 2025-08-12:
> My suggestion/proposal is to explicitly describe the IVT sharing and mention extra overhead for FUSA or/and security application.

I will add an additional note on the sharing, and mention that caution shall be given to security concerns.

In this issue, I do not see substantiated proof of a security or safety concern. I can potentially imagine a shortcoming from a security standpoint, but I have little concerns for safety. I do not see failure modes that are not detectable by typical safety mechanisms. 
E.g. if a hart accidentally uses the wrong vector table entry, this will immediately be detected by a DCLS solution (and likewise I am pretty sure an ASIL-B solution will also detect it with sufficient diagnostic coverage at that level).

christian-herber-nxp, 2025-08-12:
M-mode.
It has to be, especially as it has to be done as part of the startup routine before enabling the PMP to apply execute only permissions.

christian-herber-nxp, 2025-08-12:
> * How the M-mode startup code know the S-mode interrupt vector addresses? Does it assume the static built of the combined M/S-mode binary?

Static build of M/S binary is my default assumption.
If that does not hold, the S-mode software would have to do some kind of environment call to M-mode to ask for their vector table entries to be added. I also suspect that such a system might not be so memory/SRAM constraint, so separate tables are an option.

> * Why ACLIC do support the S-mode access to interrupt configuration (exposing APLIC-like interface to S-mode), but do not allow to configure IVT directly? Why don't we move entire IC configuration to M-mode?

We do allow IVT configured directly from S-mode, in case you invest the memory area for separate tables.
Also, the entire IC configuration can be done from M-mode.
It is however unnecessary to restrict S-mode configuration the interrupts that were delegated to it.
The only reason as to why it is more restricted on a shared memory table, is that here PMP is the access control mechanism, while on CSRs it is more fine grained.

christian-herber-nxp, 2025-08-12:
To add to this, I dug into Armv8-M:

Here, there are two vector table base addresses defined, VTOR_S and VTOR_NS for secure and non-secure software.
This is very comparable to the separation between RISC-V M- and S-mode.

Further, 

> Arm recommends that VTOR_S points to memory that is Secure and not Non-secure callable.

After giving all of this quite some consideration, I see no issue with the architectural definition as is. It allows both a very resource efficient approach, with an interleaved table, and a highly robust solution, with separate tables.

reosdev, 2025-08-15:
> However, you cannot just remove interrupt0, especially when you wan to map the major interrupts into the external interrupts. So, I think it should be exc_vector or ext_vector0.

- External interrupt 0 is not allowed neither by AIA nor by legacy architecture (PLIC).
- Major interrupt 0 is not allowed by ISA.

christian-herber-nxp, 2025-08-15:
> The values written to mcause on an NMI are implementation-defined. The high Interrupt bit of mcause
> should be set to indicate that this was an interrupt. An Exception Code of 0 is reserved to mean
> "unknown cause" and implementations that do not distinguish sources of NMIs via the mcause register
> should return 0 in the Exception Code.

Just because something is reserved, we cannot assume it will never be used.

closed 2025-08-26 by christian-herber-nxp