riscv/riscv-fast-interrupt #605: Major and external interrupts unification [proposal]

evgeniy-paltsev, 2025-09-15:
Current ACLIC spec has two interrupt types: major and external interrupts.

To achieve interrupts nesting and vectoring for both interrupt types the ACLIC spec introduces several new CSRs and extensions:
- new `mivt`, `sivt`, `meivt`, `seivt` CSRs for major and external IVTs
- new Smaiae extension to provide
  - `stopi`, `mtopi` CSRs which needed for information about priority and identity of current top major interrupt
  - `iprio0-15` iCSRs array to have configurable priority for major interrupts

We want to propose an alternative approach, which is that we unify major and external interrupts:
- In ACLIC mode all major interrupt sources are forwarded to ACLIC, according to their identity numbers. CLINT is not used at all and ACLIC becomes the only source of interrupts for M/S-mode.
- When ACLIC mode is disabled (legacy mode) major interrupts are routed to CLINT (as in RISC-V ISA). In that mode there is no configurable priority for major interrupts, no vectoring, no nesting.

The resource-constrained systems targeted by ACLIC are likely to have limited number of major local interrupt sources (M-timer, S-timer, M-software interrupts). Among all of them, only the M-timer interrupt is mandatory, according to the RISC-V ISA spec. So the resources duplication between CLINT and ACLIC is minimal.

Proposed approach has several advantages:
- we don't need multiple IVTs and `mivt`, `sivt`, `meivt`, `seivt` CSRs. Existing `xtvec` CSRs can be used for vector base.
- no need in `mtopi`, `stopi` CSRs
- no need in `iprio0-15` iCSRs array
  - we can avoid introducing of new `Smaiae` extension and drop AIA dependency
    - if we drop AIA dependency we can also drop `miph`, `siph`, `mieh`, `sieh`, `midelegh` CSRs
- we simplify interrupt trap logic
- we simplify interrupt configuration logic - all interrupts are configured the same way
- we simplify `push` logic of tpush extension
- still fully compatible with RISC-V ISA
- trivial further size reduction:
  - i.e.: application-specific IC with stricter resource constraints (and without requirement of being fully compatible to RISC-V) can implement ACLIC without CLINT at all. No ACLIC architectural changes needed for that, HW still be fully compatible with ACLIC handling mode

Since this approach offers a minimalistic ACLIC implementation, it may be more beneficial to use it as the default rather than as a separate extension.

[image]

reosdev, 2025-09-16:
I don't think this proposal introduces concepts different from AIA. It inherits and is fully compliant with APLIC direct interrupt delivery. This is much better compared to reusing AIA register names with redefined functionality in background (this does not make it compatible with AIA). In the vectored mode, any solution will not be AIA compliant in part of interrupt handling. 

Saving registers is not an intend of this proposal. The target is to get only single place to work with priorities. This allows to manage interrupt nesting at the controller side in HW (like it was done in previous ACLIC versions). This significantly simplifies the implementation of advanced latency optimizations like late preemption and tail chaining. Single source of preemption may guarantee the level information (that cannot be broken by improper SW). E.g., ACLIC in primary mode can distinguish between normal and interrupt context and provide this information to the core triggering stack switch and context push/pop (instead of relying on SW agreements and complex conditions).

christian-herber-nxp, 2025-09-18:
As several of you have previously observed, most systems will have a very limited use of major interrupts.
E.g. Zephyr uses only the software interrupt, plus the timer interrupt if you choose to.
External interrupt is used in addition if you are using the PLIC.
Additionally, interrupts like the software interrupt will tend to not be direct interrupts (in Zephyr very clearly this is an OS interrupt). Same would apply for the timer interrupt. Thus, HW-vectoring does not benefit them.

I am entertaining the idea of dropping dropping the HW-vectoring of major interrupts. They could just trap to xtvec.
This gets rid of one CSR and its associated state.

I am very skeptical of moving the current major interrupts away from the register in which they have lived for the past 10+ years.
I do not yet understand why the current proposal does not support managing interrupt nesting in HW. It would be great if you could elaborate on that.

christian-herber-nxp, 2025-09-22:
Having the nesting transparent to software is not a feature in my eyes. 
SW control over the threshold is important.
It is anyway needed for PCP.
In addition, for ISR offloading, software needs to be able to do everything that hardware does to emulate an interrupt without a physical source.