riscv/riscv-fast-interrupt #607: Corrections for 8.4 and 8.7.1/2

jklockars, 2025-09-16:
In 8.4, at the end of the "stack layout", it should say "XLEN = 64" and then "64: xpistatus (XLEN).

In 8.7.1/2, a6 (x16) is not supposed to be saved and restored, right?

By the way, isn't is rather strange in 8.4 to say
...
8: x11
4: x10
etc, rather than
4: x11
0: x10
where the number is the actual offset from some starting point?

christian-herber-nxp, 2025-09-18:
thanks for the input. When making the stack layout normative, I will fix this also.

christian-herber-nxp, 2025-09-22:
> By the way, isn't is rather strange in 8.4 to say
> ...
> 8: x11
> 4: x10
> etc, rather than
> 4: x11
> 0: x10
> where the number is the actual offset from some starting point?

the starting point is sp. with an offset of zero, you would be writing into the previous stackframe.

jklockars, 2025-09-22:
I assume you are talking about a normal RISC-V stack, which grows downwards (towards lower addresses), right?

So, you actually mean negative 4, 8, etc offsets from the original stack pointer value? As in
sw x10, -4(sp)
sw x11, -8(sp)
...
addi sp, sp, -32

Rather than the normal
addi sp, sp, -32
sw x10, 0(sp)
sw x11, 4(sp)
...
or, in case you meant the other save order
addi sp, sp, -32
sw x10, 28(sp)
sw x11, 24(sp)
...

I realize that in this particular case it would be safe to update SP last, as in the top case, but then you should  (IMHO) use negative offsets (see the top case).

In the second case above, the numbers for 8.4 would be the positive offsets from the new stack pointer value, at the lower memory address, which seems to me like an easier way to describe things.

The usual reader of this documentation is more likely, I would think, to be a software developer wondering where the register values are, rather than a hardware designer wondering where to put them. Then, would you not want to refer to something that actually exists when entering the interrupt handler - which is the updated stack pointer?

commit dca3ed4d (2025-09-22): Smtp: Fixing typo in stack definition

Closes #607