riscv/riscv-fast-interrupt #616: MSI support?

mark-honman, 2025-09-22:
It is possible to support message-signaled interrupts in the same way as the IMSIC, with the addition of 
- memory mapped MSI receiving locations as described in the the AIA
- introduce an encoding (value = 2) for xaclicsourcecfg[N].SM (source mode) that indicates that the interrupt has an MSI source. 

Setting a default of 2 for xaclicsourcefg[N].SM would result in the composite MSI-enabled ACLIC controller coming out of reset functionally identical to an IMSIC. [eidelivery does not need an ACLIC option]

APLIC is really designed to work in conjunction with an IMSIC (direct delivery to CLINT is described in AIA notes as "maybe software would want to do this" & sounds like it is more for PLIC compatability than anything else).

Therefore an MSI-enabled ACLIC provides both IMSIC compatability and APLIC connectivity (APLIC interfaced via MSI).

Note: interrupts that are configured to have an MSI source receive their priority via MSI; priority configured in xacliciprio is not used.

reosdev, 2025-09-24:
How the particular MSI can be associated with ACLIC source. There is no source ID in the message (only priority).

mark-honman, 2025-09-24:
Actually it's the other way round - my original statement that priority is in the message is incorrect, 

The message written to IMSIC's setipnum_le is just the minor interrupt identity (AIA section 3.5), and priority is determined from the minor interrupt ID (AIA 3.3). 

ACLIC sources are just minor interrupt identities, and that is how incoming MSIs and ACLIC sources can be mapped to the same interrupt file. Now there are two ways that priorities could work in an MSI-enhanced ACLIC:

1. MSIs have the same priority as their interrupt ID (law of least astonishment, it's just like AIA IMSIC behaviour). 
2. MSI interrupt gets its priority from ACLIC eiprio

mark-honman, 2025-09-24:
> If you are using this priority as identity and override it with something different (defined at ACLIC side), then APLIC priority are totally ignored. Same is about software initiated MSIs, when SW use IDs as priority and expects this order will not be broken.

Good point, so the #1  approach to priorities would be the one to go for. 
 
> If compatibility with MSI is required, then why don't we just use the IMSIC-like design for ACLIC? It is definitely less confusing.

SXaclic is actually an IMSIC-like design except that there are 2 ways in which wired interrupts are different: they need source configuration and priority (yes it took me a while to figure that out, initially I though it was intended to be APLIC-like).  

SXaclic could almost be defined as an extension to IMSIC except that IMO extensions should not delete functionality (IMSIC's memory mapped MSI address).

reosdev, 2025-09-24:
Seems like there is no clarity in difference between two prioritization schemes which makes the ACLIC (APLIC) and IMSIC be totally different:

- IMISC has priority equal to identity, unique priority for each source. Priority arbitration is as simple as MSB/LSB over bit-array. 
- ACLIC (same as APLIC) priority is orthogonal to identity, one priority shared by multiple sources. Priority arbitration is N compares of (P+N)-bits values, where N is log2(number of sources) and P is log2(number of priorities). Compared values are wider than priority field which is required for ordering the interrupts of the same priority. 

This difference makes the IMSIC instantiation be cheap and reasonable. If you are moving ACLIC/APLIC arbitration complexity to IMSIC, then this instantiation becomes very expensive (especially for virtualization).

reosdev, 2025-09-25:
> While half of the comparisons are wider than the priority field, in fact they do not all need to be (P+N) wide. A filter tree can generate a bit of IID at each comparison - indicating the winner of the comparison - so that the full IID is only output by the root of the tree. The IID is otherwise effectively encoded in the topology of the tree.

Optimizations are obviously possible (presumably for both APLIC/IMSIC prioritizations) but this does not reject the fact of significantly higher complexity for APLIC-like interrupt priorities arbitration.

reosdev, 2025-09-25:
> An ACLIC that aligns with IMSIC would internally provide logic to convert the wires to the same interrupt-file interface, so there is no need to perform an intermediate conversion to MSIs

I totally agree, the wire-to-MSI conversion logic is as simple as set bit by number. 

Do I understand correctly that you are thinking about APLIC-like frontend (holds configuration and implement wire trigger logic) delivering to the IMSIC backends (just as defied by AIA)?

christian-herber-nxp, 2025-09-27:
> Yes, that's right! The only backend enhancement needed for wired interrupts is SW-defined priority. Based on customer conversations I think the SW-defined priority overhead is small, because people are asking for control over priority of a small number of highly important interrupts - so quite likely 4-bit priorities will be a popular implementation choice for an IC with only wired interrupts (would be interesting to hear what others think about this!).

Consider the config range for already deployed IP:

- Cortex-M0+: 4 prios
- Cortex-M33/-M7: 8-256 prios

Actual usage of course depends on number of interrupt sources.
This is what copilot tells me:

[image]

[image]

4 prio bits seems to be extremely popular.

reosdev, 2025-09-29:
Based on our experience I can highlight following popular scenarios of interrupt usage:

- Simplest single privilege core, single interrupt source
- Multi-privileges core, moderate number of interrupt sources/priorities (32..256 sources, 4-32 priorities/levels)
- Domain specific core (incl. automotive/industrial), wide number of interrupt sources/priorities (256+ sources, 32+ priorities/levels)

All three are equally popular but quite different in terms of units (chips). The demand to support bigger numbers significantly grows during past few years. I think we should target all configurations including high number of interrupts/priorities/levels.

mark-honman, 2025-09-29:
Alex (and others)... in the embedded space, are you seeing customer requirements for a mix of MSI and wired peripheral interrupts?

christian-herber-nxp, 2025-12-02:
is this linked to https://github.com/riscv/riscv-fast-interrupt/pull/656?