riscv/riscv-fast-interrupt #620: Clarify interaction of xithreshold and interrupt file eithreshold.

mark-honman, 2025-09-22:
The AIA eithreshold, eip, eie, etc. indirect registers are defined within an interrupt file that is only accessible via the CSRs miselect/miregX for the machine mode interrupt file, siselct/siregX for the supervisor-mode interrupt file.

Section 5 covering Smnip and Ssnip each introduce 3 new CSRs
- mithreshold / sithreshold
- mipreemptcfg / sipreemptcfg
- mpistatus / spistatus

Is mithreshold intended to be an alias of the eithreshold register of the M-mode interrupt file, and sithreshold for S-mode? 

I can sort of see how this makes sense because mpistatus needs to be saved in a preemptible interrupt handler, and if it were an indirect CSR that would require an extra instruction to write miselect. And then if mpistatus is a direct CSR and contains field that reflect CSR values in the OIC, it would seem odd that the OIC threshold comes from an indirect CSR. 

On the other hand AIA does not provide for an M-mode threshold, and AIA mtopi is not affected by any thresholds because no threshold affects major interrupts [would that cause any problems in embedded applications - I think not, if there is a convention that major interrupts are preemptible but cannot themselves preempt - other than the externel interrupt].

 But - if mithreshold and eithreshold are different registers that would also make sense, except that then it likely becomes necessary to save eithreshold of the OIC.  

However mipreemptcfg could easily be an indirect CSR, and the AIA interrupt-file indirect CSR map has free entries either side of eithreshold.

evgeniy-paltsev, 2025-09-24:
Hi @mark-honman,

Regarding the `mithreshold` / `sithreshold` and `eithreshold` registers - these are distinct registers in the current specification.

So, for each interrupt-capable privilege level, we currently have a pair of threshold registers: the `xithreshold` CSR and the `eithreshold` iCSR. From my perspective, this duplication is unnecessary - a single threshold register per level that covers all interrupts should be sufficient.

This issue, among others, is addressed in PR #606. If we unify major and external interrupts by forwarding all major interrupts to ACLIC, then `eithreshold` will cover all interrupts, and the `xithreshold` CSRs can be removed.

mark-honman, 2025-09-30:
I've been wondering whether there is a way to make eithreshold the "one true threshold". That would avoid creating a variation vs AIA that deprecates eithreshold, and in some cases eliminate the need for mithreshold/sithreshold. 
NB this comment is "thoughts for further discussion" rather than a "yes we should do this".

The AIA iprio indirect CSRs are optional, and if not implemented then major interrupt priority-order (high to low) is mext, msw, mtimer, sext, ssw, stimer. 

So in a minimal configuration, the iprio indirect CSRs are not implemented and when the ACLIC/IMSIC reports an external interrupt that takes priority over everything else. See AIA table 5.1. In this situation mithreshold is not needed, because any of the standard major interrupts (IID < 16) should not be allowed to preempt any mext handler.

There is a bigger question of whether any other major interrupt should be able to preempt an external interrupt handler *at all*. 
AIA defines a high-priority RAS interrupt but permits it to be delivered by the IMSIC. 
Since ACLIC is a wired-interrupt handler there is no reason why other higher-priority-than-mext major interrupt cannot be wired to the ACLIC. 

It seems like there is a potential simplification if ACLIC/IMSIC interrupts can preempt major interrupt handlers, but not the other way round. Then there is no need for an explicit mithreshold threshold value.

If the hart _does_ implement AIA indirect CSRs for iprioN, a hart-level threshold is only required if major interrupts are allowed to preempt the ACLIC/IMSIC interrupt handlers. But I think this can be done in a simplistic way, which is a mipreemptcfg bit (or eithreshold bit) that indicates whether high-priority major interrupts are allowed to preempt an ACLIC interrupt handler. 

I've not thought through what should happen if there is a low-priority M-mode interrupt (e.g. mtimer) while an sext handler is running. But given that all existing mtimer handlers are written without an assumption that they will preempt, this should be a non-issue.

christian-herber-nxp, 2025-12-09:
I have run through the AIA definition of Smaia. It goes like this:

For a machine-level environment, extension Smaia encompasses all
added CSRs and all modifications to interrupt response behavior that the AIA specifies for a hart, over
all privilege levels.
[...]
Extensions Smaia and Ssaia cover only those AIA features that impact the ISA at a hart. Although the
following are described or discussed in this document as part of the AIA, they are not implied by
Smaia or Ssaia because the components are categorized as non-ISA: APLICs, IOMMUs, and any
mechanisms for initiating interprocessor interrupts apart from writing to IMSICs.
[...]
As revealed in subsequent chapters, the exact set of CSRs and behavior added by the AIA, and hence
implied by Smaia or Ssaia, depends on the base ISA’s XLEN (RV32 or RV64), on whether S-mode and
the H extension are implemented, and on whether the hart has an IMSIC.

The first two paragraphs led me to believe the IMSIC CSRs are always present when implementing Smaia.
In fact, I find the first paragraph to be quite clear bout this.
The last paragraph, however, turns this around, by stating that certain CSRs might only be implemented if the hart has an IMSIC.

In contrast to what the last paragraph states, I also cannot find the list of CSRs that are implemented when an IMSIC is present.

Anyway, I believe it is safe to assume that the CSRs defined in the IMSIC chapters are not mandatory when not implementing IMSIC.
This includes eithreshold and also eidelivery.

I will create a PR to remove both from the APLIC proposal.
eithreshold is superceeded by the global threshold CSR in the preemption extension.
Then we only have one threshold.

mark-honman, 2025-12-10:
Also from AIA, the xtopei registers are presented in the IMSIC section (3.9):
[...]
CSR mtopei interacts directly with an IMSIC’s machine-level interrupt file. 

That implies that xtopei only exist when an IMSIC is present. This also indicates that there is an existing pattern in which writes to a conventional CSR affect state that is otherwise only accessible via indirect CSRs.

Therefore in the case of thresholds, I think the cleanest solution is to keep eithreshold, and NOT create a new ithreshold CSR. 
xpistatus definitely needs to be a direct CSR. The interactions between xpistatus and eithreshold are similar to the interactions between xtopei and eip[k]. 

If we went this route then xipreemptcfg could also become an indirect CSR. 

So overall I think the ACLIC spec should:

- re-use the AIA eidelivery and eithreshold definitions, without changing them
- NOT define an additional ithreshold CSR
- describe xpinstatus as capturing the value of eithreshold at time of interrupt
- MAYBE represent xpreemptcfg.preemptmsk as an indirect register or register-field.

christian-herber-nxp, 2025-12-10:
> A what-if scenario, what-if xpintstatus was an indirect CSR?
> The immediate response would be "no", because if SW has to stack interrupted context, this adds an xiselect write to the stacking prologue or epilogue.
> However HW could stack xpistatus in a segment of indirect-CSR space (call it eistack), avoiding the need for a direct CSR. This is cheap in an FPGA implementation, but might be too much for microcontrollers in silicon?

This is a possible implementation, and the current state of the spec allows this. It would however be rather niche. It falls into the same niche as banked register files.

christian-herber-nxp, 2025-12-10:
> > Not having a global threshold seems very detrimental to what an OS will want to do.
> 
> That was stupid of me, I entirely forgot about the major interrupts. Okay AIA doesn't provide a global threshold and gets away with it, but AIA also doesn't offer preemption.
> 
> > I did not catch how you came to the conclusion that we should keep eidelivery.
> 
> Only if we keep eithreshold, then it makes sense to also keep eidelivery (unmodified) because it avoids a point of divergence from AIA.

frankly, I have no idea how the eithreshold feature is actually useful, if it does not cover major interrupts in addition.
I think that is a strange choice for AIA to not introduce the global threshold from the start.
Especially considering that IMSIC and APLIC anyway need to communicate the highest pending-enabled prio to the hart.